adafruit/Adafruit_CircuitPython_MPU6050 #3: Temperature

wallarug, 2020-01-05:
I believe the temperature formula should be:

**Option for MPU6050**
```
temp = (raw_temperature / 340.0) + 36.53
```

or **Option for MPU6500**

```
temp = (raw_temperature / 333.87) + 21.0
```

Rather than exiting formula:

```
temp = (raw_temperature + 12412.0) / 340.0
```

This is based on the code available [here](https://github.com/kriswiner/MPU6050/blob/master/STM32F401/main.cpp#L145).

I still question if the MPU6050 values are correct.  I get approximately 48 degrees C with the IMU sitting on my desk (I am using an MPU6500 which should have the same registers).   The ambient temperature is about 20 degrees C.

wallarug, 2020-01-05:
UPDATE based on datasheet:

`Temperature in degrees C = (TEMP_OUT Register Value as a signed quantity)/340 + 36.53`

FoamyGuy, 2020-01-31:
I've got a sensor on the way. I'm going to work on this once it arrives.

FoamyGuy, 2020-02-05:
This change was merged in. Thank you wallarug.

closed 2020-02-05 by FoamyGuy