adafruit/Adafruit_CircuitPython_MPU6050 #13: MpyError: Incompatible .mpy file in CircuitPython 6.0.0

marius-ne, 2021-01-18:
I'm trying to run your handy library with an Adafruit Feather M0 Microcontroller running CircuitPython 6.0.0.

I've downloaded the Adafruit Bundle with the mentioned dependencies and the MPU6050 library as the .mpy file you provided.
When importing via:
`import adafruit_mpu6050`

I get this error:
`Traceback (most recent call last):
  File "code.py", line 8, in <module>
  File "adafruit_mpu6050.py", line 56, in <module>
MpyError: Incompatible .mpy file. Please update all .mpy files. See http://adafru.it/mpy-update for more info.`

Is your library not yet compatible with CP 6.0.0 or am I doing something wrong here? 
Thank you for your help.

marius-ne, 2021-01-18:
Changed nothing, now working.

closed 2021-01-18 by marius-ne

TheKitty, 2021-01-18:
Be sure your libraries are 100% up to date when programming as they get bug fixes.

tannewt, 2021-01-20:
Note that the host OS does do some caching. The mpy may have been copied over partially before being completely written.