adafruit/Adafruit_CircuitPython_MPU6050 #14: RuntimeError - issues with device id

marius-ne, 2021-01-19:
Hello and sorry in advance if this question ist quite foolish.

I am continously running into this error-message:
`Traceback (most recent call last):
  File "code.py", line 27, in <module>
  File "adafruit_mpu6050.py", line 164, in __init__
RuntimeError: Failed to find MPU6050 - check your wiring!`

From what I can gather from the source code this is an issue with the "device-id" ("WHO_AM_I" - register in the datasheet). Now I don't know what's causing this, presumably this register is somehow different on my device?

I'm using Feather M0 and the MPU6050. I've resoldered the pins, tried adding pullups, attached an external battery, replaced the wiring but this all didn't change anything.

This is the relevant section of my code:
```
import board
import microcontroller
import busio
import adafruit_mpu6050

i2c = busio.I2C(board.SCL, board.SDA)

mpu = adafruit_mpu6050.MPU6050(i2c)
```

Have you any idea what could be causing this? Thanks in advance.

tannewt, 2021-01-20:
Please post on the Adafruit Support forum for technical support: https://forums.adafruit.com/viewforum.php?f=60

closed 2021-01-20 by tannewt

marius-ne, 2021-01-20:
Now I don't want to unnecesserarily criticize your code and maybe I'm not understanding it correctly but from what I can gather from the source code:
`_MPU6050_DEVICE_ID = 0x68  # The correct MPU6050_WHO_AM_I value`
[...]
`_MPU6050_WHO_AM_I = 0x75  # Divice ID register`

This is valid from looking at the register map. But then:
```
if self._device_id != _MPU6050_DEVICE_ID: 
    raise RuntimeError("Failed to find MPU6050 - check your wiring!")
```
With `self._device_id` being:
`_device_id = ROUnaryStruct(_MPU6050_WHO_AM_I, ">B")`

If I understand correctly you're comparing "0x68" with "0x75" and raising the error if they're not the same. 
Might this be the issue?