adafruit/Adafruit_CircuitPython_MPU6050 #18: Feature request: read from onboard DMP

John-Trager, 2021-05-27:
The 6050 MPU has a Digital Motion Processor (DMP) chip onboard that runs motion algorithms and sensor fusion to create more accurate state data. If this could be read and accessed using this package that would be great!

jposada202020, 2021-05-28:
Thanks, there are not plans that I know of to add this feature at the moment.
However, we consider any community PR contribution. Let us know if you could provide one. Thanks.

SquirtleSquadLeader, 2022-08-30:
@ladyada  I am currently using this sensor on a project.  I would like to contribute to this particular sensors CP module.    

Are there any submission guidelines for Adafruit libraries?  Or just submit working code and see if it gets accepted???  I never participated in any opensource stuff before so I am a little bit of a rookie in this department.  I appreciate any guidance!

ladyada, 2022-08-30:
yep here's a guide - https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github - if you follow those steps you will make a PR we can review for inclusion :)

SquirtleSquadLeader, 2022-08-30:
@ladyada Awesome.  Thank you.  Time to get to work.