adafruit/Adafruit_CircuitPython_MPU6050 #35: Failed to find MPU6050 but it's detected by i2cderect

Aldhanekaa, 2023-03-07:
So I'm using GY-521 MPU-6050 for my Jetson Nano, So I tried to run this code example https://github.com/adafruit/Adafruit_CircuitPython_MPU6050/blob/main/examples/mpu6050_simpletest.py then there is an error `Failed to find MPU6050 - check your wiring` . 

My wirings:
```
scl -> (28 pin)
sda -> (27 pin)
gnd -> (gnd)
vcc -> (vcc)
```

The `i2cdetect` still able to detect the i2c though.
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- 69 -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

I also change line 10 from the example to be set the address to 105 (i2cdetect detects 0x69) 
`mpu = adafruit_mpu6050.MPU6050(i2c, 105)` .

My MPU-6050 led is still on, but why it fails to find MPU6050?

revell1, 2023-03-07:
My first guess would be that when you use "i2cdetect" it is using the correct pins for SCL and SDA, but when you use
the line of code 
"i2c = board.I2C()  # uses board.SCL and board.SDA"

then that code is using different pins.
What are the values for "board.SCL" and "board.SDA"
do they match your hardware of pins 28 and 27?

Aldhanekaa, 2023-03-07:
> My first guess would be that when you use "i2cdetect" it is using the correct pins for SCL and SDA, but when you use the line of code "i2c = board.I2C() # uses board.SCL and board.SDA"
>
> then that [...]

in my code, I connect the i2c variable to board.SCL_1 and board.SDA_1
```
i2c = board.I2C(board.SCL_1, board.SDA_1)
```
why? because if I connected it to board.SCL and board.SDA the error would be `no I2C device at address: 0x69`

Aldhanekaa, 2023-03-07:
> My first guess would be that when you use "i2cdetect" it is using the correct pins for SCL and SDA, but when you use the line of code "i2c = board.I2C() # uses board.SCL and board.SDA"
>
> then that [...]

wait do we have to follow hardware pin to connect the i2c cable? isn't it enough to connect it to the i2c pin on jetson nano?

Aldhanekaa, 2023-03-07:
Do I get `Failed to find MPU6050 - check your wiring` because the MPU broken?

jposada202020, 2023-03-08:
@Aldhanekaa. Hello :). It seems to me that this is more a question than a problem with the library. I have just checked the sensor, and it worked fine using the RP2040 feather. I would recommend following this guide
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/i2c-sensors-and-devices.
I have no knowledge of how the Jetson works, but you could take a look here https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/i2c-sensors-devices.
Another suggestion would be to contact the sensor manufacturer to see if they have any more information.

Aldhanekaa, 2023-03-08:
> @Aldhanekaa. Hello :). It seems to me that this is more a question than a problem with the library. I have just checked the sensor, and it worked fine using the RP2040 feather. I would recommend [...]

I follow this tutorial https://automaticaddison.com/visualize-imu-data-using-the-mpu6050-ros-and-jetson-nano/ to run MPU6050 on jetson nano, I've followed all of the steps though even when after I changed the i2c to pin 5,3 (SCL,SDA) it still unable to find the MPU. But, is it possible this occur because my MPU broken? I think I do so because I once scratched the board with screwdriver because the soldier tin from different pin was joined (I'm newbie in soldering :D ), however if it really broken why the LED still turned on and i2cdetect still able to detect the address?.

jposada202020, 2023-03-08:
@Aldhanekaa the library is looking for a sensor in the `0x68` address, so you could change the value here to that, so it could read your sensor :). Hope that helps. Good luck! 

https://github.com/adafruit/Adafruit_CircuitPython_MPU6050/blob/a11a310006a92d57dc65547a6220ec6c9f9ab583/adafruit_mpu6050.py#L55

Regarding, the `broken` sensor, light comes from power, and not directly from the sensor, so it does not indicate that the sensor is working or not. No worries, soldering is an art :)
If you have any other question, please come by the adafruit discord channel. Maybe some people could help you :)

Aldhanekaa, 2023-03-08:
> @Aldhanekaa the library is looking for a sensor in the `0x68` address, so you could change the value here to 0x69, so it could read your sensor :). Hope that helps. Good luck!
> 
> https://github.com/adafruit/Adafruit_CircuitPython_MPU6050/blob/a11a310006a92d57dc65547a6220ec6c9f9ab583/adafruit_mpu6050.py#L55
> 
> Regarding, the `broken` sensor, light comes from power, and not directly from the sensor, so it does not indicate that the sensor is working or not. No worries, soldering is an art :) If you have any other question, please come by the adafruit discord channel. Maybe some people could help you :)

Hm still returns an error cant find MPU. 

I've changed the library python script as you suggest, however it seems like the error occurred because it tries to check.  
```python
    def __init__(self, i2c_bus: I2C, address: int = _MPU6050_DEFAULT_ADDRESS) -> None:
        self.i2c_device = i2c_device.I2CDevice(i2c_bus, address)

        print(f"{self._device_id}  {_MPU6050_DEVICE_ID}")

        if self._device_id != _MPU6050_DEVICE_ID:
            raise RuntimeError("Failed to find MPU6050 - check your wiring!")

```

and it prints `152  105`, why does the device_id returns 152?

tekktrik, 2023-03-15:
The `_device_id` get the response of asking the device to respond to a specific I2C command with it's own address.  So since it doesn't match the I2C address you provided, it doesn't seem to be communicating with it.  Is the response consistently `152`, or does it change?

Aldhanekaa, 2023-03-18:
> The `_device_id` get the response of asking the device to respond to a specific I2C command with it's own address. So since it doesn't match the I2C address you provided, it doesn't seem to be communicating with it. Is the response consistently `152`, or does it change?

The response consistently `152`.

FoamyGuy, 2024-12-20:
This seems to be an issue with the jetson nano or something else in your environment rather than this library. 

I tested with current version of the library and circuitpython on a feather S3 and confirmed the driver initilizes successfully.

closed 2024-12-20 by FoamyGuy